lsa-pucrs-old/donnie-assistive-robot-sw #107: Comando ESPIAR e COR- execução mais rápida

julianadamasio, 2017-05-29:
Boa tarde. 
Uma das reclamações dos usuários é que os comandos demoram executando. É possivel fazer eles serem mais rápidos?
Conversei com Amory e ele disse que talvez seja possivel fazer isso. Se for possivel ser mais rápido, podem encerrar a issues #106.

amamory, 2017-05-29:
Marques  e Daniel, o atraso deste comando vem do uso do GoTo ?
favor verificar se eh possivel reduzir o delay da movimentação.

edutra, 2018-10-30:
Percebi que todos os comandos estão mais rápidos. Acredito que, no momento, o que mais deixa lenta a execução dos comandos seja o Donnie falando. Se desligarmos o som, assim que a animação termina(Não está mais demorando tanto) o comando termina de executar, printando o output.