lsa-pucrs-old/donnie-assistive-robot-sw #110: donnie_player command crashes randomly on startup

rgmaidana, 2018-01-30:
Running "donnie_player" with installation from _devel_ branch randomly crashes on Stage start up.
Output on the terminal is:

`$ donnie_player`
`Executando Donnie software no diretório /opt/donnie/test/stage`
`[1] 2676`
`[2] 2684`
`[1]-  Done                    xterm -e "player ${DONNIE_EXEC_PATH}/server_donnie.cfg &> /dev/null"`
`[2]+  Done                    xterm -e "${DONNIE_PATH}/bin/alerts  &> /dev/null"`

paulasaicosque, 2018-04-03:
[image]

amamory, 2018-06-30:
I have a theory that this bug is related to the OS distribution. I use Lubuntu 14.04 in a VM and it almost never crashes. The sound also works flawless. Perhaps this bug appers in Ubuntu (Unity) and Ubuntu Mate.